pklaus/brother_ql #110: can't recognize the printer

tanqhnguyen, 2021-04-09:
Hi, 

I'm having some weird problems where `brother_ql` can't recognize the printer (I've tried with QL-700 and QL-810W) and therefore can't print.

Here is the error when I try to print something. I have confirmed that the laptop can bring normally using another tool

```
  File "/Library/Python/3.8/site-packages/brother_ql/backends/pyusb.py", line 78, in __init__
    vendor, product = int(vendor, 16), int(product, 16)
ValueError: invalid literal for int() with base 16: '0x2042_Љ'
```

and `discover` shows this

```
brother_ql --backend pyusb discover   
deprecation warning: brother_ql.devicedependent is deprecated and will be removed in a future release
INFO:brother_ql.output_helpers:  Found a label printer: usb://0x04f9:0x2042_Љ  (model: unknown)
usb://0x04f9:0x2042_Љ
```

Obviously `0x2042_Љ` is not a valid 16-base int, but I'm not sure how to fix it

Some info about the environment
Mac OS
Python 3
brother_ql@0.9.4 installed via pip 21
libusb latest version installed via brew

michaeleekk, 2021-04-12:
@tanqhnguyen I had the same issue before. I think the code couldn't print the path of the printer properly. What I did is to go to `System Information` and look for the serial number to complete the string. The result should be something like `usb://0x04f9:0x2086/000M7Z552861`

tanqhnguyen, 2021-04-13:
I have found the problem, it's because of `pyusb` version. The `get_string` (inside util) implementation of pyusb has been changed between 1.0.2 and 1.1.1. I had to force `pyusb` to be 1.0.2 for `brother_ql` to work correctly

I'm closing this issue for now

closed 2021-04-13 by tanqhnguyen