pklaus/brother_ql #127: rater image size error

devoldski, 2022-05-05:
Hello and thanks for your great software.

Im trying to print on a QL-820NWB and cant seem to get it to work, however I think i might have found out what cause this.

It seems to me that the program does not identify the correct size of the image provide.d, this might be me using the wrong commandline options, but I think I will need some helpe here.

Im trying to print an image that is prepared for a 62mm roll and use the following command

I have also tried format it for 62x100
from what i have read in the documentation the two sizes below should work.
```
info img1: 
img1|.png PNG 696x1109 696x1109+0+0 8-bit Gray 256c 62544B 0.000u 0:00.000

info img2:
img2.png PNG 696x1392 696x1392+0+0 8-bit Gray 256c 52318B 0.000u 0:00.000

Command for 62x100 (696x1109)
brother_ql -m QL-820NWB -p /dev/usb/lp0 --debug print  -lq  --red -l 62x100 img1.png

result:
DEBUG:brother_ql.raster:raster_image_size: 720x1109

Command for 62 endless   (696x1392)
brother_ql -m QL-820NWB -p /dev/usb/lp0 --debug print  -lq  -l 62red img2.png
result:
DEBUG:brother_ql.raster:raster_image_size: 720x1392

``` 

With either command i get the following debug error
DEBUG:brother_ql.raster:raster_image_size: 720x1109

To me it seems that the width is not detected properly, do anyone have a suggestion on how to rectify this?

Br
Alfa

coledeck, 2022-05-18:
What if you specify `-l 62red` instead of `-l 62x100` ?

My 820NWB prints on the DK-2251 black and red endless tape with 

`brother_ql -m QL-820NWB -b pyusb -p usb://0x04f9:0x209d/000M1Z729562 print -l 62red --red img.png`

devoldski, 2022-05-18:
Thank you. will try and see if that sorts it.