pklaus/brother_ql #128: Using 54x29 labels DK-3235 not present in label list

sawyerhammond, 2022-06-15:
There are 52x29 labels but the printer is very sensitive about what is in there and what you ask it. Returns a 'Replace media error' if I try to use the 52x29 preset so I ended up having to make my own Label() in the ALL_LABELS section in labels.py. Here is the line that I added:

Label("54x29",  ( 54,  29), FormFactor.DIE_CUT,       ( 636,  341), ( 566,  271),   80 ),

I have tested this and it works but if I try and deploy this code when it pip installs brother it will not be present. Any idea how to do that? (best case scenario this form factor is added to the library). I am using github actions that upon push installs dependencies (including brother_ql) which will not include my label and uses pyinstaller to make and publish an executable. Any ideas on this issue?

FriedrichFroebel, 2022-06-15:
I cannot see any public repositories in your account, so the installation issue is just about guessing (although probably out of scope as this most likely is a general package usage error). Have you verified that your process actually uses your patched copy instead of pulling the official release from PyPI beforehand?

sawyerhammond, 2022-06-15:
Yes, I have the modified package in my venv and it works fine for the labels I have. I think I can fork the project and keep a modified version with my source and not install brother as a dependency.

sawyerhammond, 2022-06-15:
I was able to get my test to pass, I just moved my package out of my venv and into my actual repo and removed it form my requirements.txt. Tests are now passing so I'm going to mark as closed but my suggestion to add more label sized including the DK-3235 still stands, more label sizes more labels for people to print! Big thanks to the contributors of this package, super cool!

closed 2022-06-15 by sawyerhammond