pklaus/brother_ql #136: not working on a pi?

MrGloriousFast, 2022-12-28:
Hello i cant run this on a pi. i tested it successfully on my laptop.

the udev rule on the pi:
ATTRS{idVendor}=="04f9", ATTRS{idProduct}=="209d", SYMLINK="usb/lp0", OWNER = "pi"

it create the /dev/usb/lp0 file

but when i want to run the brother-ql i get this error:

`deprecation warning: brother_ql.devicedependent is deprecated and will be removed in a future release
INFO:brother_ql.backends.helpers:Sending instructions to the printer. Total: 9544 bytes.
Traceback (most recent call last):
  File "/home/pi/.local/bin/brother_ql", line 8, in <module>
    sys.exit(cli())
  File "/usr/lib/python3/dist-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/pi/.local/lib/python3.9/site-packages/brother_ql/cli.py", line 147, in print_cmd
    send(instructions=instructions, printer_identifier=printer, backend_identifier=backend, blocking=True)
  File "/home/pi/.local/lib/python3.9/site-packages/brother_ql/backends/helpers.py", line 61, in send
    printer.write(instructions)
  File "/home/pi/.local/lib/python3.9/site-packages/brother_ql/backends/generic.py", line 32, in write
    self._write(data)
  File "/home/pi/.local/lib/python3.9/site-packages/brother_ql/backends/linux_kernel.py", line 55, in _write
    os.write(self.write_dev, data)
OSError: [Errno 22] Invalid argument
`

im using the following python script to run it:

`import subprocess

# Define the command to run
cmd = [
    "brother_ql",
    "-m",
    "QL-820NWB",
    "-p",
    "file:///dev/usb/lp0",
    "-b",
    "linux_kernel",
    "print",
    "-l",
    "29",
    "image.png"
]

# Run the command
subprocess.run(cmd)`

image.png is a normal image:

file image.png

give me this output:

image.png: PNG image data, 306 x 100, 8-bit/color RGB, non-interlaced

what can i do to print?

MrGloriousFast, 2022-12-28:
i changed the udev rule after reading another issue to:

`ATTRS{idVendor}=="04f9", ATTRS{idProduct}=="209d", SYMLINK="usb/lp0", MODE="0666"
`

still the same error

MrGloriousFast, 2022-12-29:
``i was curious what is even going on so i put the following print into the linux_kernel.py:

```
    def _write(self, data):
        print('write args:', self.write_dev, self.dev)
        os.write(self.write_dev, data)

```

and it just prints 4?

here is the output with the print:

```$ brother_ql -m QL-820NWB -p /dev/usb/lp0 -b linux_kernel print -l 29 qr.png 
deprecation warning: brother_ql.devicedependent is deprecated and will be removed in a future release
WARNING:brother_ql.conversion:Need to resize the image...
INFO:brother_ql.backends.helpers:Sending instructions to the printer. Total: 28702 bytes.
write args: 4 4
Traceback (most recent call last):
  File "/home/pi/.local/bin/brother_ql", line 8, in <module>
    sys.exit(cli())
  File "/usr/lib/python3/dist-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/pi/.local/lib/python3.9/site-packages/brother_ql/cli.py", line 147, in print_cmd
    send(instructions=instructions, printer_identifier=printer, backend_identifier=backend, blocking=True)
  File "/home/pi/.local/lib/python3.9/site-packages/brother_ql/backends/helpers.py", line 61, in send
    printer.write(instructions)
  File "/home/pi/.local/lib/python3.9/site-packages/brother_ql/backends/generic.py", line 32, in write
    self._write(data)
  File "/home/pi/.local/lib/python3.9/site-packages/brother_ql/backends/linux_kernel.py", line 56, in _write
    os.write(self.write_dev, data)
OSError: [Errno 22] Invalid argument
```

MrGloriousFast, 2022-12-29:
also i tried this:

`$ echo "asd" > /dev/usb/lp0 
-bash: echo: write error: Invalid argument
`

FriedrichFroebel, 2023-01-03:
What do you mean by the following sentence?

> it create the /dev/usb/lp0 file

`/dev/usb/lp0` is a USB device socket and no real file - you should never create it yourself. When the printer is connected, you should be able to see the active devices using `ls -l /dev/usb` - the actual device name might differ.

> and it just prints 4?

This is correct, as the backend writes to a plain file descriptor, which is identified by a simple integer.

MrGloriousFast, 2023-05-10:
Still not working. Same as above.

When i plug the printer into the usb socket /dev/usb/lp0 is there and i can see it via the ls command. if i unplug the printer the 'not real file' is not there.

What can i do? How can i start printing?

FriedrichFroebel, 2023-05-10:
Is the Editor Lite mode disabled on the device? What is the current error when you try printing to `/dev/usb/lp0` with the correct model and udev rules set up?

MrGloriousFast, 2023-05-12:
I wouldn't know how to turn on or off the 'Editor Lite mode'. Is this a Button on the printer? I looked all around and couldn't find one.

The error is the same as stated in the posts above.

Andr3wJ79, 2023-07-18:
Raspberry OS 32 bit Bullseye

usb.core.USBError: [Errno 16] Resource busy

Try remove ipp-usb: `sudo apt remove ipp-usb`

My printer rules (QL-810Wc): 

> SUBSYSTEM=="USB",ATTR{idVendor}=="04f9", ATTR{idProduct}=="209c", MODE="0666"

and a print command:

> brother_ql -m QL-810W -p usb://0x04f9:0x209c print -l 62red --red img.jpg