pklaus/brother_ql #138: "brother_ql -b pyusb discover" doesm

the-powl, 2023-02-20:


FriedrichFroebel, 2023-02-20:
Is the "Editor Lite" mode disabled through the corresponding button as per the docs?

the-powl, 2023-02-20:
yes, the "Editor Lite" mode is off

madeinaus, 2023-03-23:
Install libusb, AND use the libusb wizard to install libusb as a 'filter' into the printer USB driver chain.  pyusb is cross-platform, and can't see any USB things at all on windows: all it can see is libusb.  libusb is a platform layer, not a printer driver, and on windows, needs to be inserted into the printer-driver chain so that it can see what happens.

When you have inserted the filter, libusb can see the thing the filter is attached to, and pyusb can see the things the libusb can see.