pklaus/brother_ql #140: Print Barcode

oseiasflores, 2023-03-23:
There is a way to print an barcode code 128 using brother_ql? I trying to do with an 810w model, but seems like i only can print text or images. My process will receive de barcode value from a reader and needs to generate a barcode with that. Already done that, but only outputs text messages.

FriedrichFroebel, 2023-03-24:
What is your issue? In the easiest case, generate an image of your barcode and feed this to `brother_ql`.

oseiasflores, 2023-03-24:
> What is your issue? In the easiest case, generate an image of your barcode and feed this to `brother_ql`.

I think its related with my low knologed with printers integration... I got the impression that i needed to send the barcode "model" to printer (Some how).

I managed to create a script to do the job over network yesterday. A simple integration with https://python-barcode.readthedocs.io/en/stable/ solved my problem. 

Really thanks for your help!

closed 2023-03-24 by oseiasflores