pklaus/brother_ql #142: AttributeError exception due to backward incompatible change in Pillow 10.0.0

pieqq, 2023-07-12:
Hello! I discovered your super handy project while trying to print labels on a Brother QL-800 from Linux. Thanks a lot for the time you spent developing this solution, it's very handy!

I installed the program by following the instructions, but ran into this problem:

```
 brother_ql -b pyusb -m QL-800 -p usb://0x04f9:0x209b print -l 12 ../202306-31748.png 
deprecation warning: brother_ql.devicedependent is deprecated and will be removed in a future release
Traceback (most recent call last):
  File "/home/u/ql800/venv/bin/brother_ql", line 8, in <module>
    sys.exit(cli())
  File "/home/u/ql800/venv/lib/python3.8/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/home/u/ql800/venv/lib/python3.8/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/home/u/ql800/venv/lib/python3.8/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/u/ql800/venv/lib/python3.8/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/u/ql800/venv/lib/python3.8/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/home/u/ql800/venv/lib/python3.8/site-packages/click/decorators.py", line 34, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/u/ql800/venv/lib/python3.8/site-packages/brother_ql/cli.py", line 146, in print_cmd
    instructions = convert(qlr=qlr, **kwargs)
  File "/home/u/ql800/venv/lib/python3.8/site-packages/brother_ql/conversion.py", line 112, in convert
    im = im.resize((dots_printable[0], hsize), Image.ANTIALIAS)
AttributeError: module 'PIL.Image' has no attribute 'ANTIALIAS'
```

This is caused by Pillow 10.0.0 [which removed ANTIALIAS](https://pillow.readthedocs.io/en/stable/releasenotes/10.0.0.html#constants).

An easy fix for me was to force the installation of the previous version of Pillow, 9.5.0:

```
pip install --force-reinstall -v "Pillow==9.5.0"
```

then everything was good.

Two ways to fix this behavior:

1. Update the code (conversion.py) to use a different method as explained in the Pillow 10.0.0 changelog
2. Update the [setup.py file to fix the Pillow version](https://github.com/pklaus/brother_ql/blob/master/setup.py#L44) to 9.x.

Let me know what would be the preferred solution, I can probably help to implement this.

Cheers,

xtlc, 2023-10-09:
You can simply add:
```
from pkg_resources import parse_version
if parse_version(Image.__version__)>=parse_version('10.0.0'):
    Image.ANTIALIAS=Image.LANCZOS
```

FriedrichFroebel, 2023-10-10:
Using `pkg_resources` will only work on Python < 3.12 by default due to dropping `setuptools` from the default packages in a virtual environment. The alternative is to require and use the `packaging` package or to just replace the usage of the old way and require Python > 3.7 (Python <= 3.7 are EOL anyway).