pklaus/brother_ql #148: Cannot print with 62mm tape

taherrera, 2024-02-24:
Hello,

I can succesfully print using this command with the 29mm paper:

`brother_ql -b pyusb -p usb://0x04f9:0x209b -m QL-800 print -l 29 "fus.png"`

However, when using the 62 mm paper the printer starts flashing a red led and the sticker is not printed.

`brother_ql -b pyusb -p usb://0x04f9:0x209b -m QL-800 print -l 62 "automatico.png"`

What could be the issue ?

FriedrichFroebel, 2024-02-25:
I have no issues with the endless labels (DK-22205) on a QL-800 device, although using the Kernel backend. Are you using the correct device and have inserted the correct labels?

saurav-y, 2024-08-11:
I'm working with a Brother QL-810W
I struggled to print a label using the 62 mm reel that comes with the printer (black/red).
I tried both label types in the instructions: "62" as well as "62red". Neither worked. 

As soon as I switched the reel with a new 62 mm black only reel, the labels were printing.

giacomotaba, 2025-06-28:
I am working with a Brother QL-820NWBc
Same here. Using the sample 62mm B&R continuous label via network it does not work. My printer says "Wrong label type".
The fun fact is that using the other 29x90mm Standard Address LABEL sample it works.