pklaus/brother_ql #150: Cut after last label only

stuartlynne, 2024-03-02:
My application prints either one or two labels on pre-cut 62x100 labels.

I need to cut after the last label only.

If a single label is printed cut after it. 

If two labels are printed cut only after the second.

I am using pdf2image convert_from_bytes to generate an images list (from pdf on stdin.)

And then:
```
model = 'QL-710W'
printer = 'tcp://192.168.40.16:9100'
kwargs = { 'rotate': '90', 'cut': False, 'label': labelsize,
qlr = BrotherQLRaster(model)
instructions = convert(qlr, images, **kwargs)
send(instructions=instructions, printer_identifier=printer, backend_identifier=backend, blocking=True)
```
Setting cut to False I get no cut. Setting it to True I get a cut after each label.

Suggestions? Thanks!

FriedrichFroebel, 2024-03-02:
This does not seem to be supported without further changes as both command sets handled here are supported by your model: https://github.com/pklaus/brother_ql/blob/56cf4394ad750346c6b664821ccd7489ec140dae/brother_ql/conversion.py#L175-L187 You might try to use

```python3
convert(qlr, images, **kwargs)
qlr.cut_at_end = True
qlr.add_expanded_mode()
instructions = qlr.data
```

but I have not tested it and this probably will not actually work.

stuartlynne, 2024-03-02:
No, that did not work.

FriedrichFroebel, 2024-03-03:
Then you might want to create your custom raster class:

```python3
class CutAtEndRaster(BrotherQLRaster):
    def add_expanded_mode(self):
        self.cut_at_end = True
        super().add_expanded_mode()

qlr = CutAtEndRaster(model)
instructions = convert(qlr, images, **kwargs)
send(instructions=instructions, printer_identifier=printer, backend_identifier=backend, blocking=True)
```

stuartlynne, 2024-03-03:
That gets a cut after each label.

stuartlynne, 2024-03-03:
> Did you set cut=False for this? Then you are only left with doing the
> same as in my previous comment with add_cut_every(n) and setting n to the
> number of pages you have.

This is cuts every label if 'cut' is True, or no cuts if False

┆     kwargs = { 'rotate': '90', 'cut': True, 'label': labelsize, }
┆     qlr = BrotherQLRaster(model)
┆     convert(qlr, images, **kwargs)
┆     qlr.add_cut_every(2)
┆     instructions = qlr.data
┆     send(instructions=instructions, printer_identifier=printer,
backend_identifier=backend, blocking=True)

This was the previous. Again with 'cut' False no cuts, 'cut' True cuts
after each label.

┆     kwargs = { 'rotate': '90', 'cut': False, 'label': labelsize, }
┆     class CutAtEndRaster(BrotherQLRaster):
┆         def add_expanded_mode(self):
┆             self.cut_at_end = True
┆             super().add_expanded_mode()
┆     qlr = CutAtEndRaster(model)
┆     convert(qlr, images, **kwargs)
┆     qlr.add_expanded_mode()
┆     instructions = qlr.data
┆     send(instructions=instructions, printer_identifier=printer,
backend_identifier=backend, blocking=True)

FriedrichFroebel, 2024-03-04:
```
convert(qlr, images, **kwargs)
qlr.add_cut_every(2)
```

is wrong. You need a custom `BrotherQLRaster` class which overrides `add_cut_every` with your custom logic and `cut=True` set, as `convert` misses these parameters.

stuartlynne, 2024-03-05:
I thought something like this would work, but still get cuts after every
label:

┆     kwargs = { 'rotate': '90', 'cut': False, 'label': labelsize, }
┆     class CutAtEndRaster(BrotherQLRaster):
┆         def __init__(self, model, pages):
┆             self.pages = pages
┆             self.page_count = 0
┆             print('pages: %s' % (pages), file=sys.stderr)
┆             super(CutAtEndRaster, self).__init__(model)
┆

┆         def add_autocut(self, flag):

┆             self.page_count += 1
┆             print('add_autocut: flag: %s page_count: %s' % (flag,
self.page_count), file=sys.stderr)
┆             if self.page_count == self.pages:
┆                 print('add_autocut: calling super', file=sys.stderr)

┆                 super(CutAtEndRaster, self).add_autocut(flag)

┆
┆         def add_cut_every(self, n):
┆             print('add_cut_every: n: %s overide: %s' % (n, self.pages),
file=sys.stderr)
┆             if self.page_count == self.pages:

┆                 print('add_autocut: calling super', file=sys.stderr)

┆                 super(CutAtEndRaster, self).add_cut_every(self.pages)
┆
┆     qlr = CutAtEndRaster(model, len(images))
┆     convert(qlr, images, **kwargs)
┆     instructions = qlr.data
┆     send(instructions=instructions, printer_identifier=printer,
backend_identifier=backend, blocking=True)