pklaus/brother_ql #151: No backend avaible

LopesMarco123, 2024-03-05:
Printer QL-800

I have install lilusb and pysub, but when I run my code always get this error:

Traceback (most recent call last):
  File "c:\Users\pp01891\OneDrive - Alliance\Área de Trabalho\import matplotlib.py", line 53, in <module>
    send(instructions=instructions, printer_identifier=printer, backend_identifier=backend, blocking=True)
  File "C:\Program Files\Python312\Lib\site-packages\brother_ql\backends\helpers.py", line 57, in send
    printer = BrotherQLBackend(printer_identifier)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\brother_ql\backends\pyusb.py", line 79, in __init__
    for result in list_available_devices():
                  ^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\brother_ql\backends\pyusb.py", line 46, in list_available_devices
    printers = usb.core.find(find_all=1, custom_match=find_class(7), idVendor=0x04f9)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\usb\core.py", line 1309, in find
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available

Can anyone help me with that?

frankcaoyun, 2024-03-26:
same error here

marrold, 2024-04-03:
What command(s) are you running? You need to select the backend with `-b`

LopesMarco123, 2024-04-03:
That was not the solution.

I copy the file libusb-1.0.dll to my system32 and it work

closed 2024-04-03 by LopesMarco123