pklaus/brother_ql #155: Compromised site?

uhthomas, 2024-06-15:
The linked site [https://brother-ql.net/](https://brother-ql.net/) is just gambling? What is it supposed to be?

[image]

FriedrichFroebel, 2024-06-15:
With the package not really being maintained, it seems like the domain registration expired as well. The original content apparently provided the docs: https://web.archive.org/web/20211209014835/https://brother-ql.net/

eloydegen, 2026-08-19:
Although the website does not work, it does redirect to GitHub so it appears to in the hands of some developer again. I hope the original owner. I can't find the source code repository of the website though.