pklaus/brother_ql #157: An exception occurred with printer: invalid literal for int() with base 10: '//alprinter.xy.com'

oxhelili, 2024-07-02:
I have a printer accessed from cloudflared platform with alprinter.xy.com. I try to ping this url from my server and can reach it, even from request the status is 200.

Problem: when I try to send print instruction from my python code the above error in title is shown.
The code is as below:

        # URL of the printer service through the Cloudflared Tunnel
        url = 'https://alprinter.xy.com'

        printer = url

        qlr = BrotherQLRaster(model)
        qlr.exception_on_warning = True

        if isinstance(path, list):
            imgs = [Image.open(path_pwd + '/media/' + path1) for path1 in path]
        else:
            imgs = [Image.open(path_pwd + '/media/' + path)]

        # Converting print instructions for the Brother printer
        instructions = convert(
                qlr=qlr,
                images=imgs,    #  Takes a list of file names or PIL objects.
                label='62',
                rotate='0',    # 'Auto', '0', '90', '270'
                # threshold=100.0,    # Black and white threshold in percent.
                red=False,
                dither=False,
                compress=False,
                dpi_600=False,
                hq=True,    # False for low quality.
                cut=True
        )
       
        try:
            msg=send(instructions=instructions, printer_identifier= printer, backend_identifier=backend, blocking=True)
            print('msg', msg)
            if msg['outcome']=='error':
                messages.error(request, 'An exception occurred with printer: ' + str(msg['printer_state']['errors'][0]))    
            elif msg['outcome']=='printed':
                messages.info(request, 'QR code was printed successfully!')
    
        except Exception as error:
            # handle the exception
            logger.error(f'An exception occurred with printer: {error}')
            messages.error(request, 'An exception occurred with printer: ' + str(error)) 

Please tell me is this the wright way to send print instructions in this case?
It's urgent for me.
Thank you

FriedrichFroebel, 2024-07-02:
Please provide the full stacktrace and full executable code - at least `backend` appears to be not set.

oxhelili, 2024-07-02:
[02/Jul/2024 12:30:46] "GET /admin/store/mobile/ HTTP/1.1" 200 38187
[02/Jul/2024 12:30:47] "GET /admin/jsi18n/ HTTP/1.1" 200 3343
2024-07-02 12:30:53,799 (WARNING) - brother_ql.conversion - Need to resize the image...
2024-07-02 12:30:53,814 (ERROR) - store.admin - An exception occurred with printer: invalid literal for int() with base 10: '//alprinter.xy.com'
[02/Jul/2024 12:30:53] "POST /admin/store/mobile/ HTTP/1.1" 302 0
[02/Jul/2024 12:30:53] "GET /admin/store/mobile/ HTTP/1.1" 200 38408
[02/Jul/2024 12:30:54] "GET /admin/jsi18n/ HTTP/1.1" 200 3343

Above is the full stacktrace and I updated the executable code.
Thanks @FriedrichFroebel

FriedrichFroebel, 2024-07-02:
This is no stacktrace - this is your application log.

You probably should have a look at the CLI help or code: https://github.com/pklaus/brother_ql/blob/master/brother_ql/backends/network.py You will see that a network-based setup expects either a hostname or IP address, optionally suffixed by a port and optionally prefixed by `tcp://` (the printer itself does not support HTTP(S), but only plain TCP). This means that in your setup `http` is considered the hostname and `//alprinter.xy.com` the port.

oxhelili, 2024-07-03:
I understand that this way,  HTTP(S),  I can not access the brother printer. 

I am explaining the situation, maybe there is another way to access it. I am kindly asking you:
I have the printer connected in WLAN way, accessible with IP and I have an app deployed in a local server in different LAN from printer and shared in internet via cloudflared. 
When the app is open from any device(different LANs), I want to print in brother printer.
Currently I can do this but only in the case when server and printer are in the same LAN. 
Is it possible and supported from printer what I am asking?
Thank you

FriedrichFroebel, 2024-07-03:
I have never tried this and never used the network backend myself, but it should work in theory. You just have to ensure that you are not specifying the `http(s)://` prefix for your printer address with `brother_ql` and that there is no firewall which would block generic TCP connections on the desired port in-between.