pklaus/brother_ql #162: Brother QL-810W Printer Not Printing on Raspberry Pi

SissiFeng, 2024-10-17:
We are experiencing issues with our Brother QL-810W printer connected to a Raspberry Pi. Print jobs are sent and queued, but the printer remains idle, with no visible errors on the printer itself. Below is a summary of our setup and the troubleshooting steps we have taken so far.

**Setup Overview:**
Printer Model: Brother QL-810W
Connection: USB connection, confirmed via lsusb.
Raspberry Pi: Running 32-bit OS to avoid compatibility issues with Brother’s i386 drivers.
CUPS Configuration: Installed and running, printer added via CUPS web interface.
PPD file: brother_ql810w_printer_en.ppd (manually extracted and installed).
Filter file: /usr/lib/cups/filter/brother_lpdwrapper_ql810w confirmed present and executable.
Perl Installed: To ensure the Brother filter script runs properly.

**Current Issue:**
Jobs Queued but Not Printing:
Print jobs are successfully sent to the printer, appear in the queue, but the printer stays idle.
lp -d QL810W test.txt
request id is QL810W-11 (1 file(s))

No response any more. 

Any guidance on how to resolve the issue or further steps we could take would be appreciated!

sgbaird, 2024-10-17:
(aside: I work with @SissiFeng) it sounds like https://stackoverflow.com/a/61771673/13697228 wouldn't work either, correct?

SissiFeng, 2024-10-17:
> (aside: I work with @SissiFeng) it sounds like https://stackoverflow.com/a/61771673/13697228 wouldn't work either, correct?

Yes, we tried this method as part of our extensive troubleshooting efforts, but unfortunately, it still did not resolve the issue. The printer remains idle despite sending print jobs, suggesting an underlying communication or driver-related problem.

FriedrichFroebel, 2024-10-18:
Please note that for the `brother-ql` library, CUPS and PPD stuff is irrelevant, as it communicates with the printer directly. What you should check:

* Remove the queued CUPS jobs.
* Disable Editor Lite mode if available.
* Ensure that the correct *udev* rules are configured for the device.
* Try with different backends (*pyusb* or *linux_kernel*).

Use the `brother_ql print` command with the correct parameters to print a test image - besides the backend and printer identifier, pay attention to specify the correct label and (for labels supporting red) the red parameter.

SissiFeng, 2024-10-18:
> Please note that for the `brother-ql` library, CUPS and PPD stuff is irrelevant, as it communicates with the printer directly. What you should check:
>
> * Remove the queued CUPS jobs.
> * Disable Editor [...]

Here’s a polite and clear response:

---

Thank you very much for the detailed information. Just to clarify, I mentioned CUPS and PPD to convey that we have tried all possible methods outside of the `brother_ql` package as well. 

Among the checks you suggested:
- We’ll double-check and remove any remaining queued CUPS jobs to avoid conflicts.
- Editor Lite mode will also be verified to ensure it’s disabled.
- We’ll review the udev rules to confirm they’re properly configured for this device.

Trying the `linux_kernel` backend is a great suggestion—this seems worth exploring further, as we’ve primarily focused on `pyusb` so far. We'll also make sure to use the correct label parameters and test with the red parameter if needed. 

Thanks again for your helpful input!

southsidebrewer, 2024-10-22:
I had this same issue, and switched my Pi to Ubuntu, and it now prints. I think the main trouble is lack of package support for the PI. there is just more support for Ubuntu.

wiremanb, 2025-06-27:
I am also unable to get the PI to print to a QL-800. I am going to attempt the same test using Ubuntu on my PI.