pklaus/brother_ql #165: Printer can no longer be discovered

Soggytoast111, 2024-12-22:
Very strange issue -

I've been using the script for a few days, not much problem. I couldn't get it to work in Windows, but it was working fine in WSL Ubuntu with the USB WSL passthrough.

Today, no longer detected.  Multiple reboots of both my system and the printer didn't work. I can see the printer in Ubuntu with lsusb, but the discover command no longer shows it, and of course doesn't print. 

I spent a few hours trying to track down how this could happen but I'm out of ideas for now.

wwqwq2313, 2024-12-22:
Hello, we tried to solve the issue.

This is what we did:

This change improves the USB device detection and error handling in the pyusb backend. It adds more robust error checking when finding and configuring the USB device, and includes a retry mechanism for device detection. This should help address issues where the printer is visible to the system but not detected by the script.

You can review changes in this commit: https://github.com/wwqwq2313/pklaus-brother_ql-165/commit/b7b99a0ba9d5ece93d004ac065567be35f2a66ea.

> [!CAUTION]
> Disclaimer: The concept of solution was created by AI and you should never copy paste this code before you check the correctness of generated code. Solution might not be complete, you should use this code as an inspiration only.

---

Latta AI seeks to solve problems in open source projects as part of its mission to support developers around the world. Learn more about our mission at https://latta.ai/ourmission . If you no longer want Latta AI to attempt solving issues on your repository, you can block this account.

Soggytoast111, 2024-12-22:
Wow thanks I'll try that.